pocomane/MiSTer_Batch_Control #2: TGFX16, GB, GBC problem

kistjan, 2021-01-16:
Hello!

I love this script! Thank you for making it! It works exactly like I want it to with SNES, MD/Genesis, NES, SMS, GBA. However, I can't get GB/GBC and TGFX16 to work. It seems like it can't find the rompath, but it works perfectly with the other cores.

Thanks again!

kistjan, 2021-01-16:
By putting roms in /media/fat/Games/Gameboy, and no further subfolders it solves the issue with GB/GBC. However, this doesn't solve the issue with TURBOGRAFX16. It seems to always execute whatever is furthest down in the TGFX16 folder.

pocomane, 2021-01-17:
The last release should fix the turbografx issue.

However it is weird that you have to put the GC rom in the right directory to let it work. I can launch rom files placed anywhere on my SD. The only requirement is that the rom directory is present, but it can be empty.

pocomane, 2021-01-21:
Can you confirm that the the Turbografx issue is solved? Can I close this issue?

kistjan, 2021-02-01:
Sorry, for being late! Yes, everything seems to work. The GB/GBC pathing issue works now too. My mistake most likely! Thanks again!

closed 2021-02-02 by pocomane