pocomane/MiSTer_Batch_Control #5: Support games directories with sub-directories

Mellified, 2021-04-03:
Presently mbc looks for rom extensions for some supported systems only in the directory specified in the system_list - not subdirectories thereof. This seriously limits the functionality, since many users will be using "packs" from online sources which sort games into various subdirectories by region, developer, etc.

For example, the NES core shows no entries in the output from `mbc list_content` when .nes files are sorted into sub-directories of /media/fat/games/NES/.

Worse, manually specifying the system and rom to load fails when the ROM is in a subdirectory.

Input: `#/media/fat/misc/MiSTer_Batch_Control/mbc load_rom NES /media/fat/games/NES/4 Game Series Collections/Best-Of Lists/Top 10 Games to Play Today - Snackbar-games.com/Bionic Commando (USA).nes`

Output: `433 - error - Not supported - Can not link /media/fat/games/NES/ !!!MBC/~~~.nes -> /media/fat/games/NES/4 Game Series Collections/Best-Of Lists/Top 10 Games to Play Today - Snackbar-games.com/Bionic Commando (USA).nes
451 - error - Not supported - Can not bind the rom
459 - error - No such file or directory - Can not unbind the rom`

Mellified, 2021-04-03:
One of my co-developers noted he saw the same issue over CIFS shares - but it disappeared when running locally. Not sure why this would matter though.

pocomane, 2021-04-03:
About the rom_load error: first you have to quote the rom string at command line (e.g. mbc load_rom NES '/media/fat/game/NES/bla bla.nes'). You did not quote it in the reported command, but the error somehow got the complete path, so probably you executed a different command: can you report the exact command line? However the main error is reported by the symlink system call and it is "Not supported - Can not link ...":  is there some wierd fs configuration, like non common file system, or link in the path to the rom?  

About the list_content command: mbc should be able to launch any rom in any mounted filesystem, also if list_content does not list them. Probably I have to make this more clear in the Readme. list_content is limited to the DEFAULT game folder since scanning the whole system could be very slow. The main mbc task is launching roms, searching for them in some dirs is an easy bonus.

That said, why do you need scanning outside the default directory? If it is a common scenario, I can try to improve that feature (maybe with a configurable folder list, or a rom list cache). However if it became too complex I think I would move the feature in another tool in order to KISS.

For the CIFS share, I can not test such configuration, however with a quick search I came across some issue with the symbolic links: do you mount the CIFS share with the mfsymlinks option?

All this issues about the symlink probably can be resolved using a different method to load the roms, like the mount-based one in [WebMenu](https://github.com/pocomane/MiSTer_webmenu_package), however I want to be sure we need it before switch to it.

Mellified, 2021-04-03:
> That said, why do you need scanning outside the default directory? If it is a common scenario, I can try to improve that feature (maybe with a configurable folder list, or a rom list cache). However if it became too complex I think I would move the feature in another tool in order to KISS.

The issue is mbc is not scanning subdirectories *inside* the default directory. As I say, this is the default for "packs" that many users would have so it seems reasonable to support it. That said, yes it will take longer to scan all the subdirectories.

> For the CIFS share, I can not test such configuration, however with a quick search I came across some issue with the symbolic links: do you mount the CIFS share with the mfsymlinks option?

I am using the standard cifs_mount.sh script. It mounts remote directories to /tmp/cifs_mount/, then issues a `mount --bind` command to additionally mount the folder(s) inside the /media/fat folder as subdirectories. So in my case I'm mounting my entire /media/fat/games/ directory from my NAS.

No idea why this wouldn't work, but it is functionality essentially included with the MiSTer - and standard Linux functionality.

Mellified, 2021-04-03:
The webmenu system did work over CIFS by the way, so this seems like a likely fix.

Mellified, 2021-04-03:
> You did not quote it in the reported command, but the error somehow got the complete path, so probably you executed a different command: can you report the exact command line?

The commandline in the bash script is:
`"${mbcpath}" load_rom ${nextcore} "${nextgame}"`
Where:
```
mbcpath=/media/fat/misc/MiSTer_Batch_Control/mbc
nextcore=NES
nextgame="/media/fat/games/NES/4 Game Series Collections/Best-Of Lists/Top 10 Games to Play Today - Snackbar-games.com/Bionic Commando (USA).nes"
```

pocomane, 2021-04-08:
I implemented a mount trick similar to the webmenu one. This is still experimental so it is placed in a [dedicated branch](https://github.com/pocomane/MiSTer_Batch_Control/tree/feature-rom-mount). No release files are produced, so if you want to test it, you have to checout that branch e build it by yourself (the build script should work...)

pocomane, 2021-04-08:
Now Arcade should work too. For this system, and just for it, I had to put back the !MBC directory. It uses the mount trick, but that directory is still needed.

pocomane, 2021-04-18:
I opened a [more specific issue](https://github.com/pocomane/MiSTer_Batch_Control/issues/10). Please refer to that one for issue about CIFS mounts.

For the subdirectory listing, I will put the feature in stand-by, since linux has better tools out of the box for such purpose (e.g. find).

closed 2021-04-18 by pocomane