pocomane/MiSTer_Batch_Control #6: Compiling fails

mrchrisster, 2021-04-03:
Trying to compile my own version 
`arm-linux-gnueabihf-gcc: error: unrecognized command line option ‘-std-c99’; did you mean ‘-std=c99’?`
When I use
`arm-linux-gnueabihf-gcc -std=c99 -D_XOPEN_SOURCE=700 -o mbc mbc.c
root@osboxes:/home/osboxes/MiSTer_Batch_Control# ls -lah
total 96K
drwxrwxr-x  4 osboxes osboxes 4.0K Apr  3 02:00 .
drwxr-xr-x 22 osboxes osboxes 4.0K Apr  3 01:47 ..
drwxrwxr-x  3 osboxes osboxes 4.0K Apr  3 01:49 build
-rwxrwxr-x  1 osboxes osboxes  601 Apr  3 01:47 build.sh
drwxrwxr-x  8 osboxes osboxes 4.0K Apr  3 01:47 .git
-rwxr-xr-x  1 root    root     30K Apr  3 02:00 mbc
-rw-rw-r--  1 osboxes osboxes  31K Apr  3 01:49 mbc.c
-rw-rw-r--  1 osboxes osboxes 4.9K Apr  3 01:47 Readme.md
-rw-rw-r--  1 osboxes osboxes  282 Apr  3 01:47 .travis.yml
root@osboxes:/home/osboxes/MiSTer_Batch_Control# `
I get a binary that is half the size of the official release and doesnt work

pocomane, 2021-04-03:
Did you change the compile script? The current version contains:

arm-linux-musleabihf-gcc -std=c99 -D_XOPEN_SOURCE=700 -static -O2 -o mbc ../mbc.c ||die

so -std=c99, not -std-c99 as reported from your previous error.

Please also  note the -static flag: without it a dynamic binary is generated, that does not works on MiSTer since the rest of the code is compiled with glibc instead of muslc. Theoretically, you can use a standard gcc with glibc to make a dynamic binary, but in practice you have to pay attention to the glibc version, gcc compiler etc. This is was the primary reason why I decided to use gcc + muslc + static for such simple tools: it produces very portable binaries.

For the size, the current build.sh contains:

arm-linux-musleabihf-strip mbc ||die

that it is needed exactly to reduce the file size (dropping all the debug and linking information)

pocomane, 2021-04-03:
Sorry, I just realized you probably was talking of the single-line in the Readme. I fixed it. For the size, the strip trick is still valid.

mrchrisster, 2021-04-03:
I was indeed :)
Thanks for your awesome script btw, it adds much needed functionality to MiSTer. I'll try compiling again later. I'm working together with Mellified of integrating mbc into our Attract script ( https://github.com/mrchrisster/mister-arcade-attract/ )
We both use cifs shares with MiSTer and unfortunately mbc doesn't seem to like it. I'm just trying to confirm that webmenu doesnt have this issue but I'm also getting some weird behaviour when using cifs

mrchrisster, 2021-04-04:
Compiling works now with static flag, thanks. I'm trying to understand why 
`/media/fat/games/neogeo# mbc load_rom NEOGEO "/media/fat/games/NeoGeo/Last Resort (lresort).neo"
427 - error - No such file or directory - Can not resolve /media/fat/games/NeoGeo/Last Resort (lresort).neo
451 - error - No such file or directory - Can not bind the rom
459 - error - No such file or directory - Can not unbind the rom`
produces an error and cifs but works fine without cifs.
Looking at your source code
`  char* rpath = realpath(path, 0);` seems to be the culprit. Any idea?

pocomane, 2021-04-04:
I do not know how CIFS mount works. It seems wierd that they have issue with realpath: it is a standard posix function that resolves relative paths. You can try to disable it (rpath = path), but you could lose the ability to use relative path for roms (actually, re-reading the core, it seems just defensive programming to me, so it can be that all will work anyway without it).

However I am looking to how to implement the mount-trick used in WebMenu (see [#5](https://github.com/pocomane/MiSTer_Batch_Control/issues/5) ) that could fix  the CIFS issue. It should also avoid the creation of strange looking dirs and files.

mrchrisster, 2021-04-04:
That fixed it! CIFS shares are working fine now with `char* rpath = path;`, awesome!
The mount-trick sounds like a big improvement. Not really loving all the " !!!MBC" subfolders. Happy Easter :)

pocomane, 2021-04-08:
I close this issue. For the mount/realpath/CIFS issue we can refer to [this issue](https://github.com/pocomane/MiSTer_Batch_Control/issues/5).

closed 2021-04-08 by pocomane