pocomane/MiSTer_Batch_Control #7: mbc fails after a while

mrchrisster, 2021-04-07:
I've been running the following command all day with different systems:
`for (( ; ; )); do mbc load_rom SNES "$(find /media/fat/games/snes -type f \( -iname "*.sfc" \) | shuf -n 1)" && sleep 100; done`
It works for a couple rom changes but eventually fails, not bringing up the OSD menu anymore.
Issuing `mbc raw_seq MODOODDO` also has no effect on the core anymore. Only restart eventually fixes mbc.

pocomane, 2021-04-07:
I am able to reproduce this with a dalay of 10 seconds between the rom changes and waiting for about 10 minutes. Sadly I think this is a bug of the main MiSTer app. I think so because, you can rescue from such situation without rebooting in the following way

- kill the running mbc with ^C (or open another ssh)
- kill the MiSTer main app with 'killall MiSTer'
- re-launch the MiSTer main app with '/media/fat/MiSTer &'
- (optionally, if you want to it keep running after you disconnect, run the command 'disown')

Maybe, MAYBE, I can found a workaround: if I found a way with wich mbc can detect the hanging of Main_MiSTer, then it can be mbc itself to kill and restart it.

However, this should be reported to the [core developer](https://github.com/MiSTer-devel/Main_MiSTer), preferably finding a way to reproduce the bug without the use of external tool like mbc (I do not know how, changing roms manually for 10 minuted is a bit boring...)

pocomane, 2021-04-07:
The problem is with the core loading procedure itself, that is provided by the MiSTer app. So I was able to reproduce it without external tools. I opened an [issue](https://github.com/MiSTer-devel/Main_MiSTer/issues/379).

Thank for the report.

mrchrisster, 2021-04-07:
Thanks for looking into it, lets's hope it gets fixed!

pocomane, 2021-04-18:
It seems to be fixed in the last commit in Main_MiSTer.

closed 2021-04-18 by pocomane