pocomane/MiSTer_Batch_Control #9: Please add support dor MegaCD and Turbo Due CHD file format

mrchrisster, 2021-04-09:
Addes chd support to my local version of mbc with `{ "TURBOCD",        "EEMDDO" MBCSEQ,      "/media/fat/_Console/TurboGrafx16_",       "/media/fat/games/TGFX16-CD",    "chd", },`
Unfortunately with the new mount trick, I get the error "CD BIOS not found".

mrchrisster, 2021-04-09:
same with neogeo
`for (( ; ; )); do neo="$(find /media/fat/games/neogeo -type d \( -name *Eu* \) -prune -false -o -name '*.neo' | shuf -n 1)" && echo "Now playing: "$neo"" && mbc load_rom NEOGEO "$neo" && sleep 100; done` produces only black screen due to bios being not in root folder

pocomane, 2021-04-10:
I can not make tests until Monday, however probably the issue is that they search for bios in the same directory (or a sub-directory) that is binded. If it is the case, you could solve with a mix solution, like the one used for the arcade:

{ "TURBOCD", "EEMDDODO" MBCSEQ, "/media/fat/_Console/TurboGrafx16_", "/media/fat/games/TGFX16-CD/ !MBC", "chd" , REMOVE_EMPTY_ROM_DIR},

This practically uses the mount points but in a sub-directory , so the bios should be reachable. This make also the rom_list feature unusable, but you already know that Linux have better tools for that task :)

mrchrisster, 2021-04-10:
Great! Unfortunately there are some error messages, like
`438 - retrying the unbinding since the mount point is busy (/media/fat/games/TGFX16-CD/ !MBC/~~~.chd)` 10 times in a row but it works. That's really awesome! TGFX16-CD games are a pleasure to check out, some really nice pixel art!

pocomane, 2021-04-10:
That is just a log, at least if it does not explicetly say "Error" (it should happen at the 20th time). Probably I have to suppress the first Nth logs: the resource is busy because the code is still loading the rom!

mrchrisster, 2021-04-12:
Trying Neogeo with this logic works well but it sometimes fails.
"King of Fighters '96, The (kof96)" It failed on this rom and my bet it it's a problem with the "'". Any way to work around it? Also, is there another way to contact you by chance or do you prefer GH?

pocomane, 2021-04-14:
You are right, KOF96 does not load, and it is not related to the '. It seems that Neogeo too must be loaded with the "Mixed" system:

  { "NEOGEO",         "EEMODO" MBCSEQ,        "/media/fat/_Console/NeoGeo_",             "/media/fat/games/NeoGeo/ !MBC",       "neo", REMOVE_EMPTY_ROM_DIR, },

At this point, I think it is better to load all the system with this method. The only drawback is the temporary " !MBC" directory creation.

It is ok for me to continue to talk on GH, however if you want you can contact me on [telegram](https://t.me/abc123xx ).

pocomane, 2021-04-18:
I had switch to the mixed mode everywhere. It is still experimental but it does not have a separate branch anymore. The method is selected by means of the compile time flag USE_MOUNT_POINTS. The build script should make binaries for both the methods. Please refer to [this issue](https://github.com/pocomane/MiSTer_Batch_Control/issues/10) for further discussion.

closed 2021-04-18 by pocomane