pocomane/MiSTer_Batch_Control #11: Support for custom directories

mrchrisster, 2021-04-23:
`/media/fat/SNES# ./mbc load_rom SNES 3\ Special\ Chip\ Games/Mega\ Man\ X3\ \(Europe\).sfc`
Command fails when directory is `/media/fat/SNES` but works fine when directory is `/media/fat/Games/SNES`

pocomane, 2021-04-24:
I can load roms from ANY directory (including /media/fat/SNES), but /media/fat/games/SNES MUST exist, also if empty. I will update the documentation with such constrain.

Your specific directory (/media/fat/SNES and other "old-style" game folders in /media/fat) has an additional issue: if you have both /media/fat/SNES and an empty /media/fat/games/SNES, when you manually open the game-loading menu, the MiSTer will automatically show you the default and empty /media/fat/games/SNES. This still works, but it is a bit uncomfortable.

If you want to keep your game in /media/fat/SNES, and have the MiSTer continue to show you it as the starting directory, you have to not make any /media/fat/games/SNES directory, then you can use the CUSTOM system in mbc to load the games, e.g. :

```
MBC_CUSTOM_CORE="/media/fat/_Console/SNES_" MBC_CUSTOM_ROM_PATH="/media/fat/SNES" MBC_CUSTOM_ROM_EXT="sfc" /media/fat/misc/mbc load_rom CUSTOM /media/fat/SNES/MyRom.sfc
```

mrchrisster, 2021-04-24:
Thanks for the thorough explanation. It was an issue a user of our Super Attract Mode script reported. We resolved it by him moving the snes directory to /media/fat/games
We'll make it a requirement in the readme, that's probably the best way forward

mrchrisster, 2021-04-25:
I connected a usb drive to test if mbc will work from `usb0`. Unfortunately I can't get it to work with the most recent version.
`mbc load_rom NES /media/usb0/Games/NES/rom.nes` will not load the desired rom

pocomane, 2021-04-26:
Probably using `/media/usb0/games` as the default `mbc` directory could workaround this issue. However I think it is more clear to make a small change to the `Main_MiSTer`, that will also avoid the creation of MBC/~~~.rom dirs/files (definitely, I hope). I opened a [feature request issue](https://github.com/MiSTer-devel/Main_MiSTer/issues/389).

pocomane, 2021-05-03:
Since the request seems that will not be approved by the Main_MiSTer developer, I duplicated the code for the directory selection [1].

With this change both the `usb0` and `fat/games` issues should be solved.

---

[1] Probably this would have been necessary in any case, since the core still need to access files outside the "Sandboxed" directory (e.g. boot roms, arcade cores, etc).

closed 2021-05-10 by pocomane