pocomane/MiSTer_Batch_Control #13: Genesis extension causes errant region detection

tatsutron, 2021-06-04:
I've noticed that using `load_rom` for Genesis games will sometimes fail, since the rom link will use the `.gen` extension, causing the core to assume a U.S. release, which won't work for certain `.bin` or `.md` ROMs. Maybe the easiest solution would be to have separate "systems" for the three regions, each with their own specific extension?

pocomane, 2021-06-05:
I did not know that the region detection was (in some cases?) based on the file extension. However I split the megadrive in 3 system: GENESIS for .gen, MEGADRIVE for .md, and MEGADRIVE.BIN for .bin.

tatsutron, 2021-06-06:
Nice, thanks! (And thanks so much for creating this project, btw.)

closed 2021-06-06 by tatsutron