pocomane/MiSTer_Batch_Control #14: Is bin/cue doable?

tatsutron, 2021-06-18:
Hello! This is more a question than an issue. I noticed in the commit history that bin/cue is out since the implementation relies upon the game to load being a single file. Is there a clear path to overcome this limitation in the future?

pocomane, 2021-06-21:
Right now, the cue loading is broken because it needs all the files (cue and bins) to be linked in the destination directory, while mbc links only the one selected in the menu, i.e. the cue. Moreover, the additional files MUST have specific names, written in the cue file.

The more direct path I can think to, is:
- Make the loading procedure behave differently when a cue is selected
- Link the cue to the current target rom position
- Parse the cue extracting all the other needed files
- Link the files with the same name as the originals, in the same directory of the target rom

I am not still working on it because:
- There is a viable alternative, i.e. use the CHD
- I am busy with other project and this needs some attention to test that no other functionality is broken from the changes.

Obviously, If someone have a working patch, I will be very happy to accept a pull request :)

tatsutron, 2021-06-24:
Thanks so much for this detailed and helpful reply. (And thanks again for this project, which is the reason my project exists.)

pocomane, 2021-08-02:
I found a simpler way: I generate a new .cue with new references to the external files. This only needs a small find-and-replace function. Check the last release if you want to give it a try.

I was able to test it only with a very limited set of .cue, so probably we need to improve such functionality in the future. If non working .cue are found, it would be great to have issue reports with the problematic file attached (just the .cue is enough).

tatsutron, 2021-08-02:
That's awesome! Thanks! I'll update my project to your latest version and give this a shot in the near future.

tatsutron, 2021-09-09:
I've finally gotten around to testing this, but so far I haven't been able to get it to work. The command I'm using is `"/path/to/mbc" load_rom MEGACD.CUE "/media/fat/games/MegaCD/Sonic CD (USA)/Sonic CD (USA).cue"` (mentioning this because I wasn't sure whether the path should be to the .cue file or to the folder containing it, but I haven't been able to get it working either way).

The .cue file is as follows:

```
FILE "Sonic CD (USA) (Track 01).bin" BINARY
  TRACK 01 MODE1/2352
    INDEX 01 00:00:00
FILE "Sonic CD (USA) (Track 02).bin" BINARY
  TRACK 02 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Sonic CD (USA) (Track 03).bin" BINARY
  TRACK 03 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:02
FILE "Sonic CD (USA) (Track 04).bin" BINARY
  TRACK 04 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Sonic CD (USA) (Track 05).bin" BINARY
  TRACK 05 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Sonic CD (USA) (Track 06).bin" BINARY
  TRACK 06 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Sonic CD (USA) (Track 07).bin" BINARY
  TRACK 07 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Sonic CD (USA) (Track 08).bin" BINARY
  TRACK 08 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Sonic CD (USA) (Track 09).bin" BINARY
  TRACK 09 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:02
FILE "Sonic CD (USA) (Track 10).bin" BINARY
  TRACK 10 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:02
FILE "Sonic CD (USA) (Track 11).bin" BINARY
  TRACK 11 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Sonic CD (USA) (Track 12).bin" BINARY
  TRACK 12 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Sonic CD (USA) (Track 13).bin" BINARY
  TRACK 13 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:02
FILE "Sonic CD (USA) (Track 14).bin" BINARY
  TRACK 14 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:02
FILE "Sonic CD (USA) (Track 15).bin" BINARY
  TRACK 15 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:02
FILE "Sonic CD (USA) (Track 16).bin" BINARY
  TRACK 16 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:02
FILE "Sonic CD (USA) (Track 17).bin" BINARY
  TRACK 17 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:02
FILE "Sonic CD (USA) (Track 18).bin" BINARY
  TRACK 18 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Sonic CD (USA) (Track 19).bin" BINARY
  TRACK 19 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Sonic CD (USA) (Track 20).bin" BINARY
  TRACK 20 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Sonic CD (USA) (Track 21).bin" BINARY
  TRACK 21 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:02
FILE "Sonic CD (USA) (Track 22).bin" BINARY
  TRACK 22 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Sonic CD (USA) (Track 23).bin" BINARY
  TRACK 23 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Sonic CD (USA) (Track 24).bin" BINARY
  TRACK 24 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:02
FILE "Sonic CD (USA) (Track 25).bin" BINARY
  TRACK 25 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Sonic CD (USA) (Track 26).bin" BINARY
  TRACK 26 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:02
FILE "Sonic CD (USA) (Track 27).bin" BINARY
  TRACK 27 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Sonic CD (USA) (Track 28).bin" BINARY
  TRACK 28 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Sonic CD (USA) (Track 29).bin" BINARY
  TRACK 29 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Sonic CD (USA) (Track 30).bin" BINARY
  TRACK 30 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:02
FILE "Sonic CD (USA) (Track 31).bin" BINARY
  TRACK 31 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Sonic CD (USA) (Track 32).bin" BINARY
  TRACK 32 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Sonic CD (USA) (Track 33).bin" BINARY
  TRACK 33 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Sonic CD (USA) (Track 34).bin" BINARY
  TRACK 34 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Sonic CD (USA) (Track 35).bin" BINARY
  TRACK 35 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:02
```

pocomane, 2021-09-22:
Your `.cue` files seems to be correctly handled by `mbc`, but it fails to be loaded on my setup too. Maybe the MiSTer launches MegaCD games with some mount trick that makes relative paths not working.

If this is the case, probably I have to return to my first idea of link all the files into the !MBC folder.

pocomane, 2022-09-10:
I moved mbc to the new MGL feature of the mister. This have good chances to fix your issue with the cue.

pocomane, 2022-10-13:
I close this since in my tests it works, and no other feedback are provided.

closed 2022-10-13 by pocomane