pocomane/MiSTer_Batch_Control #15: Preserve original ROM name

InquisitiveCoder, 2021-07-26:
While using [Super Attract Mode](https://github.com/mrchrisster/MiSTer_SAM), which uses MBC to load randomly selected ROMs, I noticed that games usually boot up as if their save file didn't exist. This seems to be caused by MBC launching every game with a symlink named `~~~.${ext}`. After some testing I confirmed that save files, save states and screenshots all use `~~~` in place of the ROM's original name.

As a result, the core fails to recognize the files that the game would use when launched normally, and any files created when a game is launched through MBC could be overwritten the next time a game for the same core is launched. In theory, keeping the ROM's original name when creating would resolve these issues. As far as I can tell MBC deletes the subdirectory with the symlink after the game is loaded, so I'd expect the symlink to be at the top of the subdirectory regardless of its name.

pocomane, 2021-07-27:
With the mount/unmount method, the rom name is changed just to make more simple the file handling, so it should not be difficult to use the original names.

BUT I do not know if this enough to let the save files works. I will make some experiment in the next days.

InquisitiveCoder, 2021-07-29:
Today I changed `get_aux_rom_path` to use the ROM's name instead of `~~~.${sys->romext}` and was able to confirm that save files will load correctly. However, I'm not a C or Linux programmer so it's difficult for me to update `rom_unlink` to properly clean up the files before the next ROM is mounted.

pocomane, 2021-07-31:
Check the new release. It should keep the rom name, but I did not test yet if the save games work.

InquisitiveCoder, 2021-07-31:
Saves, save states and screenshots are all getting saved and loaded using the ROM's name as expected. Works perfectly. Thanks for the quick fix!

closed 2021-07-31 by InquisitiveCoder