pocomane/MiSTer_Batch_Control #16: MegaCD images not being released

rhester72, 2021-08-30:
When MBC is used to copy/launch a MegaCD title (I'm using CHD, not sure if that matters), when the emulator is killed, the file lock on the CHD doesn't seem to be released, resulting in breakage on later runs (in conjunction with MiSTer_SAM).  There have been a significant number of changes to Main MiSTer, Linux and the MegaCD core over the past few months and it's unclear which (if any) led to this breakage.

More interestingly, this problem does *not* appear to occur with TGFX-CD.

pocomane, 2021-09-01:
It is very strange that the issue appears with the MegaCD but not the TFGX-CD. I can not see any difference between them. I will try to update all the MiSTer stack soon and to reproduce this issue. In the meanwhile:

 - When did you downloaded mbc? (There where some changes in its internal operation about one month ago)
 - Does this happen every time you launch any MegaCD game, or it is sporadic?
 - Does it correctly load the next game if you delete the !MBC folder manually?

rhester72, 2021-09-01:
- I picked up MBC when I installed MiSTer_SAM, it was included
- Yes, it happens every time
- It works if I delete the folder manually for exactly one time :)

rhester72, 2021-09-09:
Another observation of the same issue:

https://misterfpga.org/viewtopic.php?p=32933#p32933

mrchrisster, 2021-09-10:
Can you try out one of the older mbc releases (eg version 4) to see if that makes it work?
Download:
`https://github.com/mrchrisster/MiSTer_Batch_Control/raw/master/mbc_v04`
Rename to `mbc` and replace at `/media/fat/Scripts/.MiSTer_SAM`

mrchrisster, 2021-09-10:
I did some testing and can confirm this error was introduced at
`https://github.com/pocomane/MiSTer_Batch_Control/commit/dcf0eec8829c6d33bef2d9064803553a2de35599`
The version before works correctly while the version after always loads the same rom

pocomane, 2021-09-10:
Thanks for the the analysis. The reason for the break in that commit, is that it changes where the rom are mounted, so it makes clear an issue that is present also in the previous versions: with MegaCD CHD the "Normal" umount does not work.

I not really understand why, however I found a workaround: use the special unmounting option "DETACH" a.k.a. "lazy". This is tried only if the "Normal" unmounting fails 10 times in 5 seconds.

This seems to fix the issue on my MiSTer setup. Please check if it works for you too.

mrchrisster, 2021-09-10:
Just tried working on a fix as well by defining the filename in MBC.c but it didn't work. I noticed megacd core now has an unmount option by pushing button a+b on the gamepad. Unfortunately this relaunches the core so i wasn't able to define a sequence through MBC. Anyways, great to hear you found a workaround!

mrchrisster, 2021-09-10:
Can confirm this works, thanks for the fix!

rhester72, 2021-09-13:
Looks good here as well, closing - thanks, @pocomane !

closed 2021-09-13 by rhester72