pocomane/MiSTer_Batch_Control #22: Support zip archives

tatsutron, 2021-10-20:
MiSTer seems to treat .zip files as if they were any other folder, would be awesome if MBC could support something like `mbc load_rom  ID /folder/file.zip/game.rom`

InquisitiveCoder, 2022-02-22:
I'm interested in implementing this. The main obstacle I see is that mbc seems to rely on having a predictable key sequence to load the ROM and as far as I know there's no way to read data from the OSD. Supporting single-ROM zips would be easy since it only requires pressing down and enter after the usual MBC key sequence for that core. I'm not sure there's a good way to support the general case of multiple ROMs though. MBC has no dependencies so there's no way to read the list of files in the ROM and predict how many times to press the down key. At best, the user would have to read the zip contents, sort the file and pass the index number of the ROM they're trying to load to MBC.

pocomane, 2022-02-22:
Sadly, I have found no other way to launch roms from the command line than emulating key presses to navigate the menu.

Adding a zip library dependency is not a big issue, probably we can found also one that can be statically linked with mbc (just to keep the build and distribution simple). There is also the following dependency-less option: using the "zip -sf" command through popen(), and read its stdout. But I am not fully sure we can rely on the MiSTer ordering of the zip contents.

InquisitiveCoder, 2022-02-23:
Thanks for chiming in pocomane. Even if we sort the list of files in the archive, that could still give incorrect results in the Neo Geo core since that core doesn't use the file names. Also, MiSTer's zip and unzip binaries can't deal with archives larger than 4 GB, which is why [Super Attract Mode](https://github.com/mrchrisster/MiSTer_SAM) uses [partun](https://github.com/woelper/partun) instead. That said, those are rare corner cases so I'll try to put something together with `popen`.

tatsutron, 2022-02-25:
@InquisitiveCoder you might not want to worry much about the Neo Geo core since support for `.mra` is either here or at least coming soon

InquisitiveCoder, 2022-03-02:
Sorry for the radio silence. We just had a baby and that's kept me rather busy!

I just found Sorgelig added a new feature to launch a core + game using a .mgl (MiSTer Game Launcher) file. This seems to solve the same problem as MiSTer Batch Control and supports zip loading (using the same "pathtoarchive.zip/game.rom" path syntax proposed here, even!) I need to play around with it but if it's fully-featured enough it's probably not worth the time for me to finish implementing this.

tatsutron, 2022-03-03:
Thanks for mentioning that @InquisitiveCoder! Had no idea. I notice that in RetroRGB's coverage (in the latest weekly roundup) it was mentioned that `Computer cores will most likely not work well with MGL files.` so perhaps MiSTer Batch Control will continue to be required for launching computer games.

InquisitiveCoder, 2022-03-04:
Yup. [Sorgelig plans on addressing that](https://github.com/MiSTer-devel/Main_MiSTer/issues/562) though so hopefully it won't be long before it works with every core.

pocomane, 2022-09-10:
From the [MGL doc](https://mister-devel.github.io/MkDocs_MiSTer/advanced/mgl/) it seems that only few core are not supported by MGL (for now?). So, since it is a lot cleared, I moved mbc to MGL.

When Sorgelig will add the zip support to MGL, mbc should be able to use it too (I do not know if it is already working). And since it is in his plans, I would close this issue.

InquisitiveCoder, 2022-09-28:
Thanks! That's great news. I'll take a look this weekend.

mrchrisster, 2022-09-28:
Zip support is working in mgl. That's how SAM works these days

closed 2022-10-13 by pocomane