pocomane/MiSTer_Batch_Control #24: Support new NeoGeo mra loading

mrchrisster, 2021-11-17:
It looks like we are getting .mra loading support for the NeoGeo soon. Would be great to have support in mbc for this (once it's part of update-all)
https://www.patreon.com/posts/party-58248703

tatsutron, 2021-11-18:
(This might be a dumb question but) does this require a change to MBC? I thought MiSTer had out-of-the-box support for loading *.rbf/*.mra via `/dev/MiSTer_cmd`.

pocomane, 2021-11-22:
Actually it is not very clear to me what is this ".mra loading support". If they are cores that automatically load the needed rom,  they should be very similar to the Arcade cores, and so they should already be supported.

And as @tatsutron pointed out, the support of such files in mbc is just a thin wrapper over the standard `/dev/MiSTer_cmd` functionality.

When the new .mra will be added to update-all, please report any issue here.

pocomane, 2022-09-10:
I moved mbc to the new MGL mister feature. Now, if something is supported by MGL, it should be supported by mbc too. If this is a feature not supported by MGL, I think the right thing to do is ask to add it directly to MGL.

closed 2022-10-13 by pocomane