pocomane/MiSTer_Batch_Control #26: Custom Core

MickGyver, 2022-03-25:
Not an issue per se, but could you please  explain how to setup the CUSTOM core testing and what command is needed to try it out?

pocomane, 2022-03-27:
I am not sure what you are trying to accomplish, however the following script correctly launches an adf for me:

```
#!/bin/sh
MBC_CUSTOM_CORE=/media/fat/_Computer/Minimig_
MBC_CUSTOM_SEQUENCE=EEMO
MBC_CUSTOM_FOLDER=Amiga
MBC_CUSTOM_ROM_EXT=adf
/media/fat/misc/mbc load_rom CUSTOM /media/fat/games/Amiga/WheelDriverAkiko.adf
```

Note:
1)  MBC_CUSTOM_ROM_PATH is not used, there is an error in the example in the documentation. I will fix it, the right variable to use is MBC_CUSTOM_FOLDER
2) MBC_CUSTOM_FOLDER should contain just the name of the folder inside the game directory, not the full path; so it should be "Amiga" in your case
3) The MBC_CUSTOM_SEQUENCE should contain just the sequence to open the rom rom selection menu, and not the one to select the rom. Mbc will append the remaining keys needed to select the rom.

If you are trying to select different things in the menu (e.g. to configure the core), you can use the raw_seq command.

MickGyver, 2022-03-27:
Thanks, I figured it out, I have created a pull request to add Amiga support!

closed 2022-03-27 by MickGyver