pocomane/MiSTer_Batch_Control #28: Display custom dialogs on screen

Paradox, 2022-08-03:
Not an issue, but, since you seem to know a bit about how the MiSTer works, I was wondering if there is any way to display arbitrary "OSD" messages, like you get when a file can't be found, or, like the one Jotego displays for his beta cores? I realize the ones displayed by Jotego's cores are coming from the core, but, is there any way to do that with something like your utility?

pocomane, 2022-09-10:
I opened a request to the main mister app [about this](https://github.com/MiSTer-devel/Main_MiSTer/issues/693). let's see how it goes.

Paradox, 2022-09-14:
This would be great for SAM to show information like tty2oled currently does, for those users without one. And, many other uses.

Paradox, 2023-08-15:
Wondering if anything ever came of this.