pocomane/MiSTer_Batch_Control #29: Atari 7800 not working

Pontiac76, 2022-09-02:
Running at the Mister command line, I run:

mbc load_rom ATARI7800 "/media/fat/games/ATARI7800/_Network/ASTEROID.BIN"

(With or without quotes)

The core starts up, I can see the screen go to the destination, and select the binary, but, the core just crashes.

If I manually select the same file, the ROM starts up without issue.  I'm running the app from /media/fat/Scripts/.barcoderattler

Ideas on what to check?

pocomane, 2022-09-02:
The timing can be controlled with the environment variables MBC_CORE_WAIT, MBC_KEY_WAIT, MBC_SEQUENCE_WAIT (more details in the readme). You can try to set them to slowdown the emulated input.

However from the code I see that only .a78 files are supported. The .bin support can be easily added, but I can not work on it before some days.

Meanwhile, if you want, you could try to use the CUSTOM system (details in the readme).

Pontiac76, 2022-09-02:
Looking at the source code, I can see that this application is purely slapping the keyboard for me.  It shouldn't be doing anything really whacky within the UI.  But I'm noticing some oddities when I try to slow things down.

I tried copying the ROM to .a26 and .a78 in the 7800 directory, and I get the exact same results.  Via the command line using mbc, all three variants (bin/a26/a78) when using the 7800 core fails to load, but if I go through the menu manually using the 7800 core, it works with .bin and .a26, but not a78.

When I slow the keyboard down rate to 1000 msec, I'm seeing some strange things.  Here's the video
https://youtu.be/EcKBuuhs2mk

I also went and took the keyboard rate up to 1100 and I did NOT end up on the same point.

To note, I don't have an MBC directory anywhere in my games, but, I do have Network (It's a symlink to my NAS) and the 7800 test on the fat partition.  I don't know where that !MBC directory came from.

Pontiac76, 2022-09-03:
I also tried the CUSTOM.  I got to playing with some of the other commands, and it seems that when I look at the cores available, CUSTOM points to the NES.  It used to point to the mister rbf....  Where are these values remembered between sessions?  I don't see it storing any information like that, unless it ends up in an environment variable that I'm missing....

pocomane, 2022-09-04:
The !MBC folder is created and delete by mbc itself. Practically mbc bind mounts your rom in a location that is the first of the list, so it can select the file always with the same key-press sequence.

In your video, you increased too much the time between key-press/release, so mister_main started to go down continously and selected a random folder/rom.

The mount system should work fine also with NAS and Symlink; I have not tested it, however if for NES works correctly, it should no be the issue.

Pontiac76, 2022-09-06:
It does work fine with the NAS and the NES.  Been working with that the last few days.  I installed the old 2600 core and I can get it to work with that.  There's just something not quite right with how the game launches.  I'm not sure if it's just that I've uncovered a deficiency with the 7800 core, or if there's something in the mister core that's not doing something quite right.

pocomane, 2022-09-10:
@Pontiac76 can you test the last commit?

I changed the core mechanism used to load the roms. Now mbc does not mount/link anything by itself, istead it uses the novel MGL feature of the mister. This impact all the cores, so you can give another try to the Atari 7800.

Pontiac76, 2022-09-11:
Tested, and it seems to work.  Much better, and less of a distraction when selecting the ROM.

I also tried against an SNES image and it spun up as well.  I like this aesthetic as well.

closed 2022-09-13 by pocomane