pocomane/MiSTer_Batch_Control #30: Issue launching NEO GEO games (cross hatch test)

felleg, 2022-11-14:
Info about my `mbc` setup:

```
Build timestamp: 2022-09-10/16:27:39+00:00
Build commit: 4dcd07e9ba4f18edcf916642d23a06bfe66a702d
```

When I try launching a NEO GEO rom using a command similar to this one:
```
mbc load_rom NEOGEO /media/fat/games/NEOGEO/1\ World\ A-Z/Blazing\ Star\ \(blazstar\).neo
```

I get a cross hatch test:
[image]

This neo geo game works when I try to load it directly, or via a mgl, which contains the following:
```
<mistergamedescription>
	<rbf>_Console/NeoGeo</rbf>
	<file delay="1" type="f" index="1" path="../../../../media/fat/games/NeoGeo/1 World A-Z/Blazing Star (blazstar).neo"/>
</mistergamedescription>
```

Interesting note, when I try to launch a Neo Geo game from mbc while another Neo Geo game is already playing, I get garbled graphics of the already running game. It seems the BIOS is loaded incorrectly. For example, in this image, we see what happens after trying to load Metal Slug 1 (via mbc) while Blazing Star is running (from a manual launch via UI):

[image]
[image]
(We see garbled graphics from Blazing Star)

pocomane, 2023-01-07:
Sorry, I missed this issue when it was open. Effectively there seems to be a bug when running a rom in the default path but with a different capitalization. I mean, on my machine:

```
mbc load_rom NEOGEO /media/fat/games/NEOGEO/Blazing\ Star\ \(blazstar\).neo
```

give issues similar to the reported one, while 

```
mbc load_rom NEOGEO /media/fat/games/NeoGeo/Blazing\ Star\ \(blazstar\).neo
```

correctly works [1]. Can you confirm that the latter works also for you? Just to be sure that there is nothing else wrong; I will try to fix this for any capitalization in the next days in any case.

Thanks for the report!

[1] note "NeoGeo" in the path instead of "NEOGEO"; If you use "mbc mgl_gen" command you can see that the generated path is wrong in the first case.

felleg, 2023-01-08:
Yup, I confirm that this works for me:

```
mbc load_rom NEOGEO /media/fat/games/NeoGeo/1\ World\ A-Z/Blazing\ Star\ \(blazstar\).neo
```

Strange, I never changed the capitalization for my `NEOGEO` folder. It has always been capitalized as far as I can tell.

Looking forward to the incominf fix, thanks for your help! :)

felleg, 2023-01-09:
The new fix works for me! Cheers, closing this issue.

closed 2023-01-09 by felleg