pocomane/MiSTer_Batch_Control #32: Support for launching scripts

tatsutron, 2023-01-04:
I'm guessing this was intentional and so not an issue per se, but I was wondering why the `SCRIPT` command is no longer supported in latest.

pocomane, 2023-01-05:
Launching script is not supported by MGL, and keeping all the old mount/link infrastructure only for them seems a bit an overkill to me. Moreover the feature was introduced just because originally it was "For free": I think I never used it.

However if you really need such feature, I could try to re-enable it (maybe finding some simpler trick to run them).

What is your usage scenario?

tatsutron, 2023-01-06:
Thanks for the information! And thanks for offering to look into re-enabling it, but actually I think our app supported it mainly just because it was there "for free" -- I don't think it adds a lot of value for us so I don't think it'd be worth your time.

closed 2023-01-06 by tatsutron